JamesBarwell/rpi-gpio.js #14: Event Listener gets undefined value

Aterocana, 2015-01-22:
Running your example code (copy and paste): 
```
var gpio = require('rpi-gpio'):
var pin   = 7;
var delay = 2000;
var count = 0;
var max   = 3;

gpio.on('change', function(channel, value) {
    console.log('Channel ' + channel + ' value is now ' + value);
});
gpio.setup(pin, gpio.DIR_OUT, on);

function on() {
    if (count >= max) {
        gpio.destroy(function() {
            console.log('Closed pins, now exit');
            return process.exit(0);
        });
        return;
    }

    setTimeout(function() {
        gpio.write(pin, 1, off);
        count += 1;
    }, delay);
}

function off() {
    setTimeout(function() {
        gpio.write(pin, 0, on);
    }, delay);
}
```
The led attached to pin 7 (gpio4) is blinking but the event listener on('change'...) fires with value undefined. Here's my console logs:
```
Channel 7 value is now undefined
Channel 7 value is now undefined
Closed pins, now exit
```
Note also that the event listener is fired just twice, but that should be normal since the language is asynchronous.

AdrianPetrus, 2015-04-08:
Led blinking but the value is undefined. Having the same issue. Any resolution so far ?

Console:
Channel 11 value is now undefined
Channel 11 value is now undefined
Closed pins, now exit

Aterocana, 2015-05-14:
Still nothing, but I suspect the library isn't supported anymore.

JamesBarwell, 2015-05-14:
It is supported in the sense that I will review and merge pull requests. I have recently updated the README and automated tests to help people make useful contributions that can be quickly merged.

I have other work commitments so have not been able to devote as much time as I would like to resolving outstanding issues.

Aterocana, 2015-05-14:
Glad to hear that. That's totally understandable, good luck with your commitments :)

blinger, 2015-05-21:
i'm having the same issue as well, were you able to find a fix @Aterocana

Aterocana, 2015-05-21:
Nope. Sorry.

JamesBarwell, 2015-08-05:
I suspect that this is also going to have been fixed by #23, so this should be fixed in v0.6.0.

I'll leave the issue open until someone (me if I get a chance!) confirms it, but just wanted to provide an update.

Aterocana, 2015-08-06:
I will test it soon and I'll let you know. Thanks, very appreciated.

JamesBarwell, 2018-11-09:
Hi, the example code was actually wrong here, and it was due a misunderstanding in how the listen function works. It won't listen to changes on output pins as I had in the example - it is for listening to changes on input pins.

I've updated the docs now to better illustrate this (with code that I've literally just tested on a Pi). Apologies for the confusion.

commit b6f1a796 (2018-11-10): Update voltage-cycling example in readme. Fix #14