JamesBarwell/rpi-gpio.js #15: Any interest in support for gpio-admin support?

LukeL99, 2015-02-12:
I forked your code and changed it to use [gpio-admin](https://github.com/quick2wire/quick2wire-gpio-admin) so you don't have to run your application as root. If there's enough interest, I'll go ahead and add it as a configurable option and pull request it. In the meantime, if you want to run your apps as a non-root user (any account in the gpio group), my fork is here: https://github.com/LukeL99/rpi-gpio.js

brandonmartinez, 2015-02-12:
> Reply: JamesBarwell/rpi-gpio.js <reply@reply.github.com>>

I’m literally working on the exact same thing (I think running as sudo was causing weird file lock issue, that was also causing the file watcher to fail). My fork is here: https://github.com/brandonmartinez/rpi-gpio.js/blob/master/rpi-gpio.js

LukeL99, 2015-02-12:
Haha, yup, it's basically the exact same commit with the functions in different places.

brandonmartinez, 2015-02-12:
Have you been able to run this properly, though? I’m getting weird file-in-use errors with gpio-admin.

Sincerely,
Brandon Martinez

(231) 233-3976
me@brandonmartinez.com
http://www.brandonmartinez.com/

Date: February 11, 2015 at 7:56:04 PM
Cc: Brandon Martinez <me@brandonmartinez.com>>
Subject:  Re: [rpi-gpio.js] Any interest in support for gpio-admin support? (#15)

LukeL99, 2015-02-12:
I haven't tested input yet (I'm just controlling relays), so no, I haven't checked. I do know outputs are functioning properly, but this could be related to #2. I'm going to check into it at some point, but I'm not at the point where I need to yet. Next phase of the project, I'll be bit-banging SPI so I'll definitely need input then.

JamesBarwell, 2015-02-12:
There was also a previous pull request here that seemed to achieve this in a slightly different way: https://github.com/JamesBarwell/rpi-gpio.js/pull/10
It was closed by the author however, and I was never sure whether it was ever fully working. The timing method it used didn't look entirely reliable, and there were no tests to prove it.

I'd really like a solution for running this module without root. It would be nice not to have to depend on gpio-admin, but if that's the only/best way to do it then yes there certainly would be interest in having a switch to allow users that choice.

@LukeL99: Looking at your patch, it would be nice to use the debug module for the output as well rather than console.log, just so that users have full control over this module's output.

LukeL99, 2015-02-12:
Seems like @brandonmartinez and I are doing a lot of the same work in parallel. Let's put together a strategy for some of these design decisions and @JamesBarwell can decide what he wants or doesn't want to merge.

1) @JamesBarwell, how attached to your indentation standards are you? I have been trying to stick to them as closely as possible, but it looks like #17 has changed it around to his indentation preferences.

2) API structure - I'm trying to change the API as little as possible. My first thought for adding config to the module was to have the user invoke the library like: `var gpio = require('rpi-gpio')({gpioAdmin: true});`, however this requires the version without gpio-admin to be initialized like so: `var gpio = require('rpi-gpio')()`, which would break existing applications. I thought about exporting an additional constructor that could be initialized like `var gpio = require('rpi-gpio').gpioAdmin()`. This would work without breaking existing applications. The third option is the one @brandonmartinez is using, where you set the GPIO library. I like this option, however it seems as though an option that you should never change once you set it should be set on initialization, not afterwards. It may be our best bet, however.

3) Testing - I'd say @brandonmartinez seems closer to finishing this feature addition than I do, but I think it should be resubmitted with the testing added for new features and passing on Travis. Besides that, good work. Looking forward to using this library in more rpi projects in the future.

brandonmartinez, 2015-02-12:
As an additional note, I haven't been able to get it working with a normal install of gpio-admin. It looks like the edge mode needs to be set to properly monitor for changes, so Wiring Pi works a lot better. gpio-admin can still be use, but we'd have to manually add in writing to the /edge file.

brandonmartinez, 2015-02-12:
Also, I'll see if I can get tests passing in my codebase later this week

JamesBarwell, 2015-02-20:
@Luke99, Regarding coding standards I think the best strategy (with any project) is to respect and stay consistent with what's already there (unless there's a good reason to change). I can sort out pull-requests to match if need be however.

With regard to setting options, I see your point, but for now I would just set them after the module has been instantiated, similar to the existing setMode method (which I would presume people would only ever call once). I think this is simple for users, keeps it similar to most other modules, and maintains backwards-compatibility.

Apologies if I'm slow with responses at the moment. I'm in the middle of a large project launch so don't have the time I'd like to devote to side projects. If people choose to fork then I will look to unify changes back in the future.

brandonmartinez, 2015-02-20:
As a note, I probably won't be submitting any additional pull requests after the one I've already submitted. Feel free to still use it, but I've taken this project in a different direction from the original purpose (almost a blend of this and onoff's approach, object-oriented style and what-not).

I definitely plan on crediting you, @JamesBarwell, but I may end up making this its own library since the overall idea is different now.

JamesBarwell, 2015-02-20:
That's fine, thanks for the contribution anyway.

JamesBarwell, 2017-04-17:
A new version `0.8.1` has been released which no longer requires root permissions, so I will finally close this issue as there's now no need to support gpio-admin. Cheers.

closed 2017-04-17 by JamesBarwell