JamesBarwell/rpi-gpio.js #19: Error using Raspberr PI 2

simar88, 2015-04-13:
I'm trying to use this module on my new Raspberry Pi 2 and i have this error:
<code>
Error: Pin has not been exported
  at /home/pi/Projects/nodejs/node_modules/rpi-gpio/rpi-gpio.js:205:20
  at process._tickCallback (node.js:355:11)
  at Function.Module.runMain (module.js:503:11)
  at startup (node.js:129:16)
  at node.js:814:3
</code>

I'm trying top open a pin and then write to it using a a custom callback in this way. 
<code>
var gpio = require('rpi-gpio');
gpio.setup(pin,gpio.DIR_OUT,write(pin,false))
</code>
I'm using:
<b>sudo comand;
node version: 0.12.1;
</b>

JamesBarwell, 2015-04-13:
Your ```write``` function is executing immediately because you have put ```(pin,false)``` after it. You need to declare ```write``` as its own callback function, then call ```gpio.write``` from within that. See "Setup and write to a pin" in the readme.

closed 2015-04-13 by JamesBarwell

simar88, 2015-04-14:
Thanks @JamesBarwell  for the fast reply!
I followed your guidelines for "setup and write" before write the issue and the function write is declared before as function that works as callback passing to it the pin and the value to write.
After some hour to test, update and upgrade raspberry i found that the problem was a sleep module for node.js. This module stop the execution of the callback so this write, that came after changing direction of the pin, doesn't fin the pin exported.
I solved before using nested operations ad then using async module to make operations in series and it works. Thanks again! :)
Using it

jgaut, 2015-11-29:
Hello,

May you put your solved code here please ?

julienvincent, 2015-11-29:
Working code would be:

```
var gpio = require('rpi-gpio');
gpio.setup(pin,gpio.DIR_OUT,write.bind(this, pin,false))
```

alexcroox, 2016-03-14:
I'm having the same issue on rPi 2 using

```
gpio.setup(17, gpio.DIR_OUT, function() {

     gpio.write(17, true, function(err) {
        if (err) throw err;
        console.log('Written to pin');
    });
});
```

```
Error: Pin has not been exported for write
```

alexcroox, 2016-03-14:
Ignore me, wasn't following the useful pin guide

[image]