JamesBarwell/rpi-gpio.js #22: fs has no method 'exists' ?

GPrathap, 2015-06-06:
This is the stack trace of error. 
/home/pi/node_modules/rpi-gpio/rpi-gpio.js:376
    fs.exists(PATH + '/gpio' + pin, function(exists) {
       ^
TypeError: Object #<Object> has no method 'exists'
    at isExported (/home/pi/node_modules/rpi-gpio/rpi-gpio.js:376:8)
    at /home/pi/node_modules/rpi-gpio/rpi-gpio.js:160:17
    at /home/pi/node_modules/rpi-gpio/node_modules/async/lib/async.js:641:34
    at Array.0 (/home/pi/node_modules/rpi-gpio/node_modules/async/lib/async.js:557:34)
    at EventEmitter._tickCallback (node.js:190:39)

GPrathap, 2015-06-06:
I resolved this error after installing newest version of NodeJs (https://learn.adafruit.com/node-embedded-development/installing-node-dot-js)

JamesBarwell, 2015-07-23:
Yes it sounds like a node version issue. Thanks for the link - I've updated the nodejs installation guidelines in the readme.

commit d887f157 (2015-07-23): Update node installation guidelines. Fixes #22
closed 2015-07-23 by JamesBarwell