JamesBarwell/rpi-gpio.js #24: pin turn off during the 'read' operation.

roccomuso, 2015-06-29:
This is my code:

exports.get = function(pin){ // perchè il file functions.js è incluso esternamente
    
	gpio.setup(pin, gpio.DIR_IN, read_state);
    
	function read_state() {
			gpio.read(pin, function(err, value) {
				var state = (value) ? "on": "off"; // The current state of the pin (true: on or false: off)
				console.log("The value is: "+state);
				return {response_code: 200, new_state: stato, path: "/toggle"};
			});
	}
	gpio.destroy();
};

When i call exports.get(11) it actually turns off the gpio 11 and print 'the value is: off'.

Why is turning off the Pin? o.O

JamesBarwell, 2015-08-06:
The code example is buggy, in that you're running ```setup``` and then immeditely (before ```read_state``` is called) running ```destroy```.

But I'm going to leave this open and have a go at re-creating it because I still wouldn't expect it to turn the pin off.

JamesBarwell, 2017-04-16:
I can't re-create this. After running the (fixed) code, it sets up with a value 'on'. The read operation doesn't change its state.

The new version 0.8.0 of the module allows you to explicitly set the pin into a high or low state so perhaps you may want to use that in order to setup the pin in the desired state.

closed 2017-04-16 by JamesBarwell