JamesBarwell/rpi-gpio.js #25: gpio.setup with gpio.DIR_OUT turns the pin on by default

roowilliams, 2015-07-07:
Initializing the pin also switches it high, is this desired behaviour?

Bacce, 2015-07-21:
Came here with the same question, I use rpi a+ and when I setup pins to DIR_OUT it set them high.
I think it should be low on setup, or at least an option to set the default value.

JamesBarwell, 2015-07-23:
It's not documented behaviour, so yes this is a bug. I can't imagine that many users would find it desirable behaviour either.

I'm confident from the test suite that the module doesn't cause a write to ```/sys/class/gpio/gpioX/value``` when ```setup()``` is run. So I'm puzzled as to what could be causing it. I'm wondering if it's perhaps an underlying bug in the OS code.

julienvincent, 2015-07-23:
A fix would be to write `0` to  `gpioX/value` Within the export function. I can include this in my fork

JamesBarwell, 2015-07-23:
Do we really want to default the pins during setup though? It seems like a very heavy-handed fix. It's not clear from the bug report but I'm assuming that this happens with DIR_IN as well as DIR_OUT, and it would definitely not be expected for that.

I think the best approach here would be to see if the same behaviour occurs when writing to the filesystem manually. At least then we could rule out whether it's a bug in this module or something lower down.

julienvincent, 2015-07-23:
You are right - probably best to find out where this behavior is coming from.

andres-torres-marroquin, 2015-10-08:
I'm using a Raspberry Pi v2 (jessie), initially I thought this library was working wrongly with a Sainsmart relay (those relays work with active low), and I got into this issue.

After using a multimeter and testing the library, when you `gpio.setup(pin, gpio.DIR_OUT)` I read 0v with the multimeter, so it is working as intended. But I do find benefits on allowing the user to setup the way it works, using /sys/class/gpio/gpioX/active_low, which after some research and experimenting, I found out that it inverts the way it works.

Example:

```bash
echo 23 > /sys/class/gpio/export
echo 'out' > /sys/class/gpio/gpio23/direction # (multimeter reads 0v)
echo 1 > /sys/class/gpio/gpio23/value # (multimeter reads 3.3v)
echo 1 > /sys/class/gpio/gpio23/active_low
echo 1 > /sys/class/gpio/gpio23/value # (multimeter reads 0v)
echo 0 > /sys/class/gpio/gpio23/value # (multimeter reads 3.3v)
```

Do you think that different revisions of the RPI work different? I have a RPI A+ that I could try with.

mike-engel, 2016-01-15:
Hello! So, in my case, I have a relay tied to the GPIO that is *backwards*, in that `HIGH` = off, and `LOW` = on. Very weird but being able to specify a default state to `HIGH` (so it's "off") would be super nice.

macguru2000, 2016-03-14:
I am having the same issue, if it is an underlying OS problem can anyone help me figure out how to fix it? I can assure that when I use bash to send either `out`, `low`, or `high` to the exported gpio pin direction, it always triggers my relay immediately... very problematic.

noelhibbard, 2016-05-20:
I have the Sainsmart relay board so this patch comes in handy. Thanks macguru2000.

h3smith, 2016-12-27:
I assume this project is dead, but this can be accomplished. The command necessary is doing the gpio export. When you do export, you pass the direction as 'high' or 'low' not just 'out'. 'high' or 'low' is the same as 'out' but specifies the starting value.

JamesBarwell, 2017-04-16:
The options to pass high and low are included in the latest version, `0.8.0`, using the patch from macguru2000. This issue should now be resolved. Cheers.

closed 2017-04-16 by JamesBarwell