JamesBarwell/rpi-gpio.js #26: 'gpio.on' binding not working on PI2

hadesara, 2015-08-01:
I am trying out this library and I am not unable to follow the examples since the callback functions to gpio.on 'change' and 'export' are not being triggered. I notice that RPI2 is not mentioned anywhere on this repo. Do you support RPI2 ?

julienvincent, 2015-08-01:
The RPi 2 is supported, however in the current master branch of this module, change listeners do not fire. We are busy implementing a new change listener which should be coming to master quite soon. So long you can use [my working fork](https://github.com/julienvincent/rpi-gpio.js) which is hardware tested and functioning.

julienvincent, 2015-08-03:
This issue has been fixed.

hadesara, 2015-08-03:
Awesome! Thank you. Referring to your latest comment, Is it merged into the repo's current master? Or do I still need to pick you fork?

JamesBarwell, 2015-08-03:
Published today as 0.6.0, so just pull it in through npm.

closed 2015-08-05 by JamesBarwell

hadesara, 2015-08-05:
Sorry, haven't been able to test this yet. I will test it soon and provide feedback if any.

hadesara, 2015-08-09:
Negative, Things seems to be broken 0.6.0. The change listeners do not work and even the callback read listener returns 'undefined'. Where as reverting to v0.5.2  has at least the callback read listeners works correctly.

hadesara, 2015-08-09:
I take this back, the issue was with access permissions to /sys/class/gpio/gpio21/edge. Looks like all is well. Thank you for this update!

julienvincent, 2015-08-09:
Ah yes this is something I intend to work on when I get time. Having to `sudo` run everything is impractical.