JamesBarwell/rpi-gpio.js #27: Unable to install on Node 4.0.0

natdm, 2015-10-03:
I'm getting a node-gyp-rebuild error while installing rpi-gpio on node 4.0.0.

Command: pi@raspberrypi ~/node_projects/gpio $ sudo npm install -g --unsafe-perm rpi-gpio

Any idea what's going on? I was told to try the unsafe-perm and it should help. Here's the output from that:

http://pastebin.com/MkfWUnbY

julienvincent, 2015-10-03:
@natdm This system has not yet been thoroughly tested on node -v`4`, however after giving it a test run on node 4, I was unable to reproduce. Can you post your full log please

natdm, 2015-10-03:
There's nothing in /var/log/messages as of recent. All I have right now is the pastebin link in the original. Wonder if my node install is botched?

julienvincent, 2015-10-03:
Try using node `-v4.1.1` which is the current stable. If that is still giving you issues, open an issue on the epoll module as that seems to be what is causing your issue.

JamesBarwell, 2015-10-05:
This looks like the error I was getting when trying to get the tests to pass on v0.4. I had to update the test runner to use G++ v4.8 to handle the compilation of epoll.

The commit here fixed the CI builds: https://github.com/JamesBarwell/rpi-gpio.js/commit/fa1b7aa9168c680a9bcc677ed6ba3dc01c15beac
The failing builds before the commit: https://travis-ci.org/JamesBarwell/rpi-gpio.js/jobs/81541424

I've not tried running it on the Raspberry Pi yet, but I would imagine that if you can set it up like I have for CI, then that should work. Perhaps something like this:
```
sudo apt-get install ubuntu-toolchain-r-test g++-4.8
CXX=g++-4.8 npm install
```

closed 2015-11-12 by JamesBarwell

julienvincent, 2015-11-17:
I can now confirm this. node-gyp is having trouble compiling. `ubuntu-toolchain-r-test` is not a available package for the raspberry pi. I will see what installing `g++-4.8` does.

julienvincent, 2015-11-17:
**Solved:** @JamesBarwell you were actually right - after installing `g++-4.8` you need to tell the system to use that version. [Here](https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons) is a link on how to do this. I will add it to the readme.