JamesBarwell/rpi-gpio.js #33: Error while installing

jfarago, 2016-01-16:
pi@raspberrypi:~/aquarium_monitor $ npm install rpi-gpio
\
> epoll@0.1.16 install /home/pi/aquarium_monitor/node_modules/rpi-gpio/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/aquarium_monitor/node_modules/rpi-gpio/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../src/epoll.cc:15:0:
../node_modules/nan/nan.h:328:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
epoll.target.mk:84: recipe for target 'Release/obj.target/epoll/src/epoll.o' failed
make: *** [Release/obj.target/epoll/src/epoll.o] Error 1
make: Leaving directory '/home/pi/aquarium_monitor/node_modules/rpi-gpio/node_modules/epoll/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.1.13+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/aquarium_monitor/node_modules/rpi-gpio/node_modules/epoll
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

npm ERR! epoll@0.1.16 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the epoll@0.1.16 install script.
npm ERR! This is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.1.13+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "rpi-gpio"
npm ERR! cwd /home/pi/aquarium_monitor
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/aquarium_monitor/npm-debug.log
npm ERR! not ok code 0
pi@raspberrypi:~/aquarium_monitor $

julienvincent, 2016-01-16:
Please refer to the (readme)[https://github.com/JamesBarwell/rpi-gpio.js/blob/master/README.md#dependency]

jfarago, 2016-01-16:
I looked at that an verified i was running gcc/g++ -v4.8 or higher.

julienvincent, 2016-01-16:
What version of node are you running?

julienvincent, 2016-01-16:
I would suggest updating to the latest version of node.

attacking, 2016-04-07:
very very very thanks for your rpi !   @julienvincent

julienvincent, 2016-04-07:
@attacking No problem.

closed 2016-04-28 by JamesBarwell