JamesBarwell/rpi-gpio.js #37: read pin & blocking

guizmo51, 2016-05-30:
Hi, thanks for your very useful library ! 
I try to listen changes on a specif pin (on input only). It works if I use "gpio.read()" but the script is waiting for action after.. How to solve it ? 

My goal is to listen chane but the eventlistener is not working for me :( 

thanks for your help !

julienvincent, 2016-05-30:
Hi, please can you include some code to show what you are trying to do.

guizmo51, 2016-05-30:
Hi,
Here is what I have : https://gist.github.com/guizmo51/84f0881d1a5634c58ae22f86772dadb8 (blocking after the console.log() )

Intially I used the "change" listener but it doesn't work.

julienvincent, 2016-05-30:
Are you sure it's not just `hanging`. Can you continue code execution? It is expected to hang after a pin setup as from that point on it is listening for changes.

guizmo51, 2016-05-30:
Sorry for the missunderstood. Yes it's probably hanging, but hanging about what ? I just want to listen changes on this pin.. do you have working exemple ?

julienvincent, 2016-05-30:
after a setup, the module will be start polling for interrupts signalling a change. When a pin change happens, the module will emit a change `event` you can hook in and listen to these events by adding a call back function like so:

```javascript
gpio.on('change', (channel, value) => {
   // channel -> pin that changed
   // value -> the new state of the pin. 0/1
})
```

guizmo51, 2016-05-30:
Well, I already try that but no change was détected (instead of the getter). I will try again tonight. Thanks !

julienvincent, 2016-05-30:
The reason it was not detecting a change is because you are not specifying the `edge`. If you look at the [documentation](https://github.com/JamesBarwell/rpi-gpio.js#setupchannel--direction-edge-callback) you will see that you need to specify an `edge` when listening for changes.

julienvincent, 2016-05-30:
In future please make sure to read documentation before opening an issue.

guizmo51, 2016-05-30:
Hi Julien, 
I understand to read the documentation before and this is why I did. During the day I hadn't my Pi. Toninght I just try again the same snippet and now it works. Thanks.

closed 2016-05-30 by guizmo51