JamesBarwell/rpi-gpio.js #38: Pin has not been exported for write

SimonTheEngineer, 2016-07-22:
Hi, I've just started using rpi-gpio and wanted to test the functions. I'm trying to write to a pin (7) but I'm getting the following error:

Pin has not been exported for write

Node: v6.3.0
Raspbian: 8.0 Jessie

If the project is still active, any advice would be great.
Thanks.

julienvincent, 2016-07-22:
Hi, could you please share some of your code so as to make it easier to see where things are going wrong.

SimonTheEngineer, 2016-07-22:
Damn yeah sorry. I'm away from the computer at the moment but I just copied the example code.

kirgy, 2016-08-02:
Same issue, using copy-paste  example code:
```
var gpio = require('rpi-gpio');
 
gpio.setup(7, gpio.DIR_OUT, write);
 
function write() {
    gpio.write(7, true, function(err) {
        if (err) throw err;
        console.log('Written to pin');
    });
}
```

```
pi@raspberrypi:~/test $ node -v
v4.2.1
```

btinoco, 2016-08-20:
I am having the same issue, details:

+ nodejs v6.2.2
+ npm v3.9.5

Code:
```
    var gpio = require('rpi-gpio');                        
    var pin = 25;                                          
                                                        
    gpio.setup(pin, gpio.DIR_OUT, writeToPin);             
                                                             
    function writeToPin(){                                 
                                                           
        gpio.write(pin, true, function(err){           
                                                           
            console.log(err);                      
                                                         
        });                                            
     }                                                      
```

Error:
```
    raspberrypi ~/nodegpio $ npm start

    > nodegpio@1.0.0 start /home/pi/nodegpio
    > sudo node index.js

    Error: Pin has not been exported for write    at /home/pi/nodegpio/node_modules/rpi-gpio/rpi- gpio.js:227:20
    at _combinedTickCallback (internal/process/next_tick.js:67:7)
    at process._tickCallback (internal/process/next_tick.js:98:9)
```

JamesBarwell, 2016-08-30:
I think that these may be the same issue as was reported and solved in #41. I suspect an error is being thrown after `setup` and is not being handled here. The error may well be due to a lack of root permission when running the script. Could you try running again using sudo, and if that fails then ensure you are catching errors after setup?

```
  gpio.setup(pin, gpio.DIR_OUT, writeToPin);             

    function writeToPin(err){  
      if (err) throw err;

      gpio.write(pin, true, function(err){
```

mxyue, 2016-10-08:
Error: Pin has not been exported for write

maybe you need start program with root or sudo permission
like sudo npm start

miopa, 2016-12-08:
I had the same problem.  It turned out - I was writing to a pin that can't be used for writing. The confusion originates in the different pin numbering schemes. The library uses WiringPi as default scheme, while most of the references online use the BCM naming.

Here you can find both schemes referenced: http://ocw.cs.pub.ro/courses/iot/labs/04

yunw, 2017-04-17:
I resolved this issue after applying both fix from @miopa and @JamesBarwell

JamesBarwell, 2017-04-17:
Glad you got it fixed. Just to let you know, there's a new version of the module `0.8.1` which shouldn't require root permission anymore.

closed 2017-04-17 by JamesBarwell