JamesBarwell/rpi-gpio.js #39: gpio.EDGE_BOTH EIO: i/o error, write

reinvantveer, 2016-07-23:
rpi-gpio throws on setting up pin 7 for listening:

```js
'use strict';

const bunyan = require('bunyan');
const log = bunyan.createLogger({ name: 'RPI button app' });
const gpio = require('rpi-gpio');
const express = require('express');
const app = express();

gpio.on('change', (channel, value) => {
	log.info('Channel ' + channel + ' value is now ' + value);
});

gpio.setup(7, gpio.DIR_OUT, gpio.EDGE_BOTH, err => {
	if (err) throw err;
});
```
Gives me:
```
pi@raspberrypi ~/Git/rpi-button-app $ sudo nodemon ./index.js 
/home/pi/Git/rpi-button-app/index.js:14
	if (err) throw err;
	         ^

Error: EIO: i/o error, write
    at Error (native)
```

Running Raspbian wheezy, node v4.4.7 on a v1 B Pi.
I could just set up a time interval, as the normal read and write are working fine. But I rather like the evented feature of rpi-gpio :)

Thanks for any pointers

julienvincent, 2016-07-23:
Hi, the error you are experiencing is due to a badly configured/broken environment. It is a hardware related problem that will need to be resolved on the pi you are running on.

reinvantveer, 2016-07-23:
Alright, with that much tongue in your avatar I'll take you word for it :)
I did a restart on my Pi but to no avail :(
But for my use case, it's fine to go without evented stuff from rpi-gpio.

Thanks for the quick reply!

closed 2016-07-23 by reinvantveer