JamesBarwell/rpi-gpio.js #41: [Error: Pin has not been exported]

hems, 2016-08-27:
I'm trying to setup/read using this library but i'm getting "Pin has not been exported" error, here is my code:
````
gpio.setup 23, gpio.DIR_IN, gpio.EDGE_BOTH, ( error, data ) ->

  gpio.read 23, ( error, data ) ->

    if error then return console.log 'error reading', error

    console.log data
````

here is the error:
````
error reading [Error: Pin has not been exported]
````

if i manually export and try to read via gpio it works //=
````
echo 23 > /sys/class/gpio/export
gpio read 23
0
gpio read 23
1
````

rPi 2, model B

hems, 2016-08-27:
It turns out it was a "permission denied" error comming from the method setEdge.

Since the error wasn't being handled on the "setup" function the next block wasn't being executed and i also didn't realise the problem.

Actually the library just failed silently, when logging the error i got
````
{ [Error: EACCES: permission denied, open '/sys/class/gpio/gpio11/edge']
  errno: -13,
  code: 'EACCES',
  syscall: 'open',
  path: '/sys/class/gpio/gpio11/edge' }
````

the solution was to run my script with "sudo" or give the right permissions to my user

JamesBarwell, 2016-08-30:
Thanks for the report and for the solution.

I think there are two issues here:
1) the README examples are not very good as they don't catch the errors on pin setup, and people are following their bad example.
2) the API is perhaps a bit too granular for how people want to use the module. Setting up and writing to a pin must be the most common use-case, yet it requires two method calls and error handling at each level. Perhaps there should be a simpler API that combines the step and provides a single place where error-handling is required.

hems, 2016-09-01:
@JamesBarwell thanks a lot for your library and for your input. i wish i had done a pull request instead of simply opening a issue. the problem is that i'm quite new to dealing with gpio so i'm backtracked on my project trying to figure out some of my application problems first - apologises about that.

Also, on a similar subject, if you don't mind me asking here:

I'm trying to figure out is how to read encoders, i have seem some examples using C and it seems they read a "stream" that comes from the encoder..

I'm not sure how we could make this happen using the "read" function here? 

I tried to read as fast as possible, but it didn't seem to be right way of doing?

Perhaps for encoders and other sort of inputs we could somehow integrate with node.js streams making some sort of "socket connection" to the gpio ?

And last but not least, i believe perhaps we should replace using async for something like [bluebird Promises](http://bluebirdjs.com/docs/getting-started.html) ? 

I believe it would make it easier to handle the errors and also to read the source code.

Again, thank you very very much for putting this together.

Hopefully i'll figure my stuff out soon and get back with some pull requests instead of feature requests!

JamesBarwell, 2016-09-19:
I'd be interested in streaming input, but I'm afraid it's not something I have much knowledge of at the moment, nor will I have any time to look at. This module really just wraps the existing GPIO filesystem functions, so it may be a bit out of scope.

There was a request to use promises in: #34. I'm not ruling it out but it's not the route I want to go down at the moment. The API could definitely be improved, but I am still keen to stick to the style of the node.js API at present.

alexandrbig, 2016-11-01:
Hi. Is there any update on this topic?
I am facing the same issue with setting up the edge. And for me it is quite confusing. I am running script on Raspbian Jessie with pi user, which already added to gpio group, but no success dealing with npm module. 
I don't want run script with root permissions.

JamesBarwell, 2016-11-01:
I've only done a little reading on this so I'm prepared to be convinced otherwise, but from what I can see root permissions are required to write to the filesystem for these pins. This module is really just a wrapper for those filesystem operations so it's not within scope of this module to try and work around it. I don't think this module should be involved in messing around with file permissions or user groups, etc.

So at the moment my thinking is that it's up to the user of this module to either compartmentalise their application appropriately, so that only some of it is running with root permissions, and/or do something with the sudoers file to restrict what can get run. There's a blog post along similar lines here: http://www.mobilewill.us/2012/07/raspberry-pi-and-gpio-permissions.html

alexandrbig, 2016-11-01:
@JamesBarwell thank you for the reply. I was just wondering if you have solution for that.
Now I launched my app with service with root user and it works properly.
Good luck in research.

hems, 2016-11-03:
@JamesBarwell / @alexandrbig  i'm pretty sure you don't need to be root, just google it and you will find! unfortunately i don't have the link as reference right now

awong-dev, 2016-12-25:
I think I found the root cause for this. Wrote it up in #48 .

JamesBarwell, 2017-04-17:
There's a new version `0.8.1` which should no longer require root permissions.

closed 2017-04-17 by JamesBarwell

PiyushRamuka, 2017-04-19:
I am using 0.8.1 and still having this issue on my Pi 3B. I am doing something wrong?

JamesBarwell, 2017-04-20:
@PiyushRamuka Could you raise a new issue and include the code you're using + all the details of the error please.

PiyushRamuka, 2017-04-20:
@JamesBarwell  - created one. Thanks!
[Done](https://github.com/JamesBarwell/rpi-gpio.js/issues/59)