JamesBarwell/rpi-gpio.js #42: read pins value not right

mxyue, 2016-10-15:
`
var lPins = [31,33,35,37]
lPins.forEach(function(channel){
            gpio.read(channel, function(err, value) {
                    console.log(channel + 'The value is ' + value);
            });
})`

--------
31The value is true
33The value is true
35The value is true
37The value is true
--------
31The value is false
33The value is false
35The value is false
37The value is false
--------
31The value is true
33The value is true
35The value is true
37The value is true

--------
31The value is false
33The value is false
35The value is false
37The value is false
--------
31The value is true
33The value is true
35The value is true
37The value is true

JamesBarwell, 2016-10-20:
There is a problem with your code. `setup` is an asynchronous method, and you need to wait for it to complete before attempting to `read`. At the moment your code is setting up four pins, then immediately trying to read all four. There's no defined behaviour for this scenario.

I'd suggest having a look at the `async` node module to help with the control flow you're trying to achieve.

mxyue, 2016-10-25:
@JamesBarwell thanks for your help

closed 2016-10-25 by mxyue