JamesBarwell/rpi-gpio.js #44: Need a way to specify use of internal pullup/pulldown resistors for GPIO pins

2sk21, 2016-11-05:
There does not seem to be any way to specify the use of the internal pulp/pulldown resistors in the code - or have I missed something? I did see a way to do this in the Python RPi library.

seymar, 2017-01-12:
Same problem here

JamesBarwell, 2017-01-13:
This is already being discussed in here: https://github.com/JamesBarwell/rpi-gpio.js/issues/25

There is an outstanding PR (https://github.com/JamesBarwell/rpi-gpio.js/pull/35) that should address this in a forthcoming version.

closed 2017-01-13 by JamesBarwell

pwnate, 2017-05-27:
I don't believe #25 or PR #35 address this issue. The internal pullup/pulldown resistors are for **direction in** and not **direction out**. The issue/PR address initializing the *direction out*.
There's a thread about this [here](https://raspberrypi.stackexchange.com/questions/62654/pull-up-resistors-from-rpi-gpio-js).

JamesBarwell, 2017-06-07:
Apologies, I had quite a few issues to close off a result of landing that patch, and I misunderstood this when I gave my previous response. Thanks for the link to the stackexchange discussion.

I think, given what nmaas87 says on that discussion, this isn't something that should be in-scope for this module. nmaas87 is correct that this module is just providing bindings around the `/sys` path that controls the pins. If it can't be done through there then it's not something I want to support.