JamesBarwell/rpi-gpio.js #46: Error: Cannot find module 'rpi-gpio'

TheNotary, 2016-11-11:
I'm probably overlooking something obvious, but I can't seem to get this code working.  Is there a naming issue I'm not seeing?

Installing:
```
$ npm install -g rpi-gpio

> epoll@0.1.20 install /home/pi/.npm-packages/lib/node_modules/rpi-gpio/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/.npm-packages/lib/node_modules/rpi-gpio/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/.npm-packages/lib/node_modules/rpi-gpio/node_modules/epoll/build'
/home/pi/.npm-packages/lib
└─┬ rpi-gpio@0.7.0 
  ├── async@1.5.2 
  ├─┬ debug@2.3.2 
  │ └── ms@0.7.2 
  └─┬ epoll@0.1.20 
    ├── bindings@1.2.1 
    └── nan@2.4.0 
```

Using:
```
$ node
> var gpio = require('rpi-gpio');
Error: Cannot find module 'rpi-gpio'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at repl:1:12
    at sigintHandlersWrap (vm.js:22:35)
    at sigintHandlersWrap (vm.js:96:12)
    at ContextifyScript.Script.runInThisContext (vm.js:21:12)
    at REPLServer.defaultEval (repl.js:313:29)
    at bound (domain.js:280:14)
> gpio
undefined
```

JamesBarwell, 2016-11-14:
I've not seen this before. I suspect it's because you've installed it globally and should just be doing it locally.

```
npm install rpi-gpio
```

TheNotary, 2016-11-14:
Thanks @JamesBarwell, it works without issue when installed locally and when node is run from the folder with the `node_modules/` folder, this can be closed.  Also, I'm noticing by setting the NODE_PATH variable, I can require in packages globally, my system was slightly misconfigured.  e.g. `export NODE_PATH=/home/pi/.npm-packages/lib/node_modules:$NODE_PATH`

closed 2016-11-14 by TheNotary