JamesBarwell/rpi-gpio.js #50: Maintainer stagnation

julienvincent, 2016-12-27:
It's beginning to look like this repository is no longer actively maintained, with issues and a couple good PR's popping up.

@JamesBarwell I understand you may be busy/unable to actively maintain this library, but this is quite a popular library and is continuously gaining followers. This inactivity is hurting the repo.

I'd like to see this repository grow and be improved, but in order for that to happen - something needs to change.

fivdi, 2017-03-08:
There are of course alternatives, some of which are very good even if I may say so myself :)

buchannon, 2017-03-08:
Just curious what some alternatives are?

I would also like to see the current rpi-gpio repo continue to grow as I'm using it for a project. 
Does the owner just need to give permission for others to commit into it?

fivdi, 2017-03-08:
> Just curious what some alternatives are?

An alternative offering the same functionality packed a little differently would be [onoff](https://github.com/fivdi/onoff). It runs on the Pi and a number of other platforms and doesn't require root/sudo permissions on the Pi.

A second alternative which supports not only blazing fast digital IO but also PWM, servo control, alerts (timers), low latency interrupt handlers, banked GPIO, trigger pulse generation and pull up/down resistor configuration would be [pigpio](https://github.com/fivdi/pigpio).

> I would also like to see the current rpi-gpio repo continue to grow as I'm using it for a project.
> Does the owner just need to give permission for others to commit into it?

This is also a potential alternative.

JamesBarwell, 2017-03-09:
I've already offered that to Julien, since he contributed a good chunk of code and done a lot to assist with the issues, but like me he's also very busy.

I have tried to make time for reviewing and feeding back on pull requests. I'm still waiting on the outstanding ones to get fixed up. I just can't necessarily devote time to adding features and fixing bugs myself, at the moment. I do intend to spend more time on this in the future but I have to put paid work and real life first.

julienvincent, 2017-04-18:
Its great to see an increase in activity! Thanks @JamesBarwell and everyone participating for the good work!

closed 2017-04-18 by julienvincent