JamesBarwell/rpi-gpio.js #53: Cannot open pin for read

javirs85, 2017-03-12:
Im running raspi Model B Rev 2 512MB, I installed the module and succesfully opened a pin for writing on it. Succesfully made a LED blink.
Then I try to open a port for reading and I get and error 

> { Error: EINVAL: invalid argument, write errno: -22, code: 'EINVAL', syscall: 'write' }

This the relevant part of the code. 

```
var gpio = require('rpi-gpio');
server.listen(8080, function() {
	console.log('Servidor corriendo en http://localhost:8080');
	
	gpio.setup(7, gpio.DIR_OUT, control);
	gpio.setup(6, gpio.DIR_IN, control);
});

function control(err)
{
	if(err)
		console.log("Control function: " + err);
	else
		console.log('ok');
}
```

Any clue ??

javirs85, 2017-03-12:
Ok, error was that pin 6 i GND. GPIO6 is actually mapped to pin 22. Setting 22 solved the isue.

Is there a way to name pins in a form other than the physical pin? I´m using a custom expansion board and this is hell :(

thecodershome, 2017-03-12:
There is a fork of this repo that has MODE_RAW. https://github.com/timelapseplus/node-linux-gpio

javirs85, 2017-03-13:
sorry, I'm new to github, how can I choose to install the forked version?

and.. what's the mode RAW ? in the link u sent me there is nothing about it.. :(

thecodershome, 2017-03-13:
To install the forked repo run `npm install timelapseplus/node-linux-gpio` or `npm install timelapseplus/node-linux-gpio --save` to add the repo to your dependencies.
Since the package name was changed you will need to use `var gpio = require('linux-gpio')` instead of `var gpio = require('rpi-gpio')`.

If you use `gpio.setMode(gpio.MODE_RAW)` you can "attempt" to use any channel.

```js
function getPinRaw(channel) {
        return channel + '';
    };
```

javirs85, 2017-03-13:
can you gimme an example of "channel" ?

thecodershome, 2017-03-13:
`gpio.setup(7, gpio.DIR_OUT)`

Take a look at this, https://pinout.xyz/pinout/pin3_gpio2
It should help with picking channels.

thecodershome, 2017-03-13:
By default this library uses the physical pin values, `MODE_RPI`.

javirs85, 2017-03-14:
My pi is 1 rev 2, I only have 26 pins so the graph is not fitting my setup. Anyhow, what's the raw name I have to enter to access GPIO6 ?