JamesBarwell/rpi-gpio.js #55: Allow Setup to take in an array of channels?

ShayD16, 2017-04-03:


JamesBarwell, 2017-04-16:
Yep, sounds like a good idea. There's some work in progress at the moment to improve the API generally so this might be something that fits in nicely after that work is merged.

thecodershome, 2017-04-18:
This could be added with the PR #54 as `gpio.promise.setupAll` and could look like the example below. However if a promise rejects then `setupAll` will reject. 
```js
    /**
     * @param {number[]}    channels
     * @param direction
     * @param edge
     * @returns {Promise}
     */
    setupAll: function (channels, direction, edge) {
        if (!Array.isArray(channels))
            return Promise.reject(new Error('Channels must be an array'));

        return Promise.all(
            channels.map(function (channel) {
                return GPIO.promise.setup(channel, direction, edge);
            }))
    }
```
A possible alternative would be to catch errors inside `setupAll` and return an array of successful channels.

```js
setupAll: function (channels, direction, edge) {
        if (!Array.isArray(channels))
            return Promise.reject(new Error('Channel must be an array'));

        // Iterate over channel
        return Promise.all(
            channels.map(function (channel) {
                return GPIO.promise.setup(channel, direction, edge)
                    .then(function () { // return channel
                        return channel;
                    }, function (e) { // ignore error
                    });
            })) // returns array of results
            .then(function (chanList) {
                return chanList.filter(Number); // filter out 'undefined'
            })
    },
```

```js
var gpio = require('rpi-gpio.js'),
    gpio_p = gpio.promise;

var list = gpio_p.setup([3, 5, 7, 99])  // rejects 99
list.then((res) => {
    console.log(res) // [3, 5, 7]
})

```

JamesBarwell, 2017-04-18:
My thoughts on this:
- the Promises API should just mirror the existing one for the time being, at least until it has been well-used. So I'd prefer to work out how to add it to the standard API then wrap it with the promise interface if necessary.
- having the whole promise reject / an error being throw if just a single pin in the setup command fails feels like the more correct behaviour to me. Else we will end up with silent failures that inevitably go unhandled and cause other weird behaviour. The error could contain info about the state of the pins.

thecodershome, 2017-04-18:
Here is what I've come with for `setupAll` on the standard API https://github.com/thecodershome/rpi-gpio.js/commit/e56c96bd5373ed7e41d17d213469d932923b9dc8

```js
var gpio = require('rpi-gpio.js');

function done(err, res) {

/*
  err = [{channel: 99, error: Error}]; // Error with stack
  res = [{channel: 3, result: 'result from setup'},
    {channel: 5, result: 'result from setup'},
    {channel: 6, result: 'result from setup'}];
*/
}

var list = gpio.setupAll([3, 5, 7, 99], done)
```

Currently result is undefined, but if we ever return with something its avaliable

ShayD16, 2019-01-09:
Is this being worked on? I'd like to have a look at this if it isn't already done/in progress.

ShayD16, 2019-01-09:
Is this being worked on currently? I'd like to have a go at it if it isn't already done/in progress?
PS: Sorry for accidentally closing the issue

JamesBarwell, 2019-01-09:
Hi, there has been some work on this from @thecodershome above which is worth looking at. I haven't picked it up myself.

I'm unsure how useful this is now that the promise API is available, and my current thoughts are that anyone who wants to deal with multiple operations in parallel will be drawn towards using that, and will use `setup()` and `Promise.all()`. Have you got a specific problem or example that would help to illustrate where this would be useful?

ShayD16, 2019-01-10:
Hey, nothing specific I was just tinkering around when I realised I could only open one channel at a time, I just realised my original issue gets completely solved by using Promise.all().