JamesBarwell/rpi-gpio.js #59: Running without root permission

PiyushRamuka, 2017-04-20:
I am using v0.8.1 and i get following error when I try to run this application without root permission. It works perfectly with root.

Error: Pin has not been exported for write
    at /var/www/smartlyteplus/node_modules/rpi-gpio/rpi-gpio.js:243:20
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickCallback (internal/process/next_tick.js:104:9)

Here is my code
```javascript
var gpioDriver = require('rpi-gpio');

function setPin(pin, state, cb){
    gpioDriver.setup(pin, gpioDriver.DIR_OUT, function(){
        gpioDriver.write(pin, state, function(err){
            if(err){
                return cb(err);
            }
            return cb();
        });
    });
}
```

thecodershome, 2017-04-21:
Looks like user permissions. I created a standard user without access to sudo. I recieved the same message 

Error: Pin has not been exported for write
    at /home/tester/testpi/node_modules/rpi-gpio/rpi-gpio.js:243:20
    at _combinedTickCallback (internal/process/next_tick.js:67:7)
    at process._tickCallback (internal/process/next_tick.js:98:9)

I also opened node in command line and ran these commands
```js
var gpio = require('rpi-gpio')

 gpio.setup(7,function(err,res){console.log(err)})
```

```js
{ Error: EACCES: permission denied, open '/sys/class/gpio/export' at Error (native)
  errno: -13,
  code: 'EACCES',
  syscall: 'open',
  path: '/sys/class/gpio/export' 
}
```

So I figured out that a user needs to be in the `gpio` group. 
To see the current user groups run `groups` into the terminal.
If ther don't have `gpio` then run this to add them `sudo adduser USERNAME gpio` where USERNAME is the name of the user.
After that everything worked just as it should.

paulostergaard, 2018-03-23:
Adding homebridge user to the gpio group worked (finally!) for me as well.  Good tip.

JamesBarwell, 2018-04-11:
Thanks all for the bug report and for the solution. I'll get this added to the readme.

commit 5af31f1f (2018-04-12): Add non-sudo instructions. Fixes #59
closed 2018-04-12 by JamesBarwell