JamesBarwell/rpi-gpio.js #61: Node module version out of date

Booligoosh, 2017-04-29:
Hi there,
When I try to use this module I get this error:

```
Uncaught Error: The module '/home/pi/rubiks-robot/node_modules/epoll/build/Release/epoll.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 51. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at bindings (/home/pi/rubiks-robot/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/pi/rubiks-robot/node_modules/epoll/epoll.js:1:192)
```

Any idea what this means or how to fix it?

Thanks

JamesBarwell, 2017-05-11:
I've not seen it before. It's a problem with the epoll dependency rather than this module. Which version of node are you using?

JamesBarwell, 2018-11-09:
All dependencies have been updated so I don't think this is going to be relevant anymore. Please open a new issue if it continues to occur. Cheers.

closed 2018-11-09 by JamesBarwell