JamesBarwell/rpi-gpio.js #66: Is there a way with the GPIO utility when exporting a pin to DIR OUT, it comes in high value?

lwaclawiak, 2017-06-26:
Is there a way with the GPIO utility when exporting a pin to DIR OUT, it comes in high value?

I saw that this has been fixed for the rpi-gpio.js library, but with the utility this remains unsolved.

I program my scripts with bash, and use a bank of relays that are fired as soon as I export the pins as out.

The same thing happens if I send by bash the value "out" to the direction file of the pin.

Thanks!

pricemi115, 2019-04-26:
This bug seems quite old (not not a problem....or no longer a problem, anyway). I just started playing with rpi-gpio and do the same thing without a problem in the current version. For anyone else having this same question, simply use DIR_HIGH as the direction. DIR_OUT and DIR_LOW are equivalent.

JamesBarwell, 2019-04-30:
Thanks for following up on this. You're right, it was probably broken or undocumented when this was raised. I'll close it now as it seems clear in the documentation that this is how it works.

closed 2019-04-30 by JamesBarwell