JamesBarwell/rpi-gpio.js #67: Error: Module did not self-register.

eaglejs, 2017-06-26:
I am trying to require rpi-gpio.

I am running node 8.0.0 and npm 5.0.3: I have downgraded node 6.3.0 and npm to 3.x and still get the same result. when I try to run the application and for some reason I get the following stacktrace:

`Error: Module did not self-register.
    at Object.Module._extensions..node (module.js:598:18)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
    at bindings (/Users/jeagle/repos/gate-keeper/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/Users/jeagle/repos/gate-keeper/node_modules/epoll/epoll.js:1:99)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/Users/jeagle/repos/gate-keeper/node_modules/rpi-gpio/rpi-gpio.js:6:20)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Module.require (module.js:513:17)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! gate-keeper@1.0.0 start-be: `node ./src/service/app.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the gate-keeper@1.0.0 start-be script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.`

If I comment out the require, the application works as intended minus the GPIO features. :) I might be missing something so please request any additional information and I'll gladly provide it. Thank you all very much for your time.

eenewbsauce, 2017-07-05:
Are you on a Mac?

eaglejs, 2017-07-05:
Yes, I am on a mac, that might be the reason it won't install... I would think it would still install so I can develop the application? Do you have any suggestions to substitute? I was thinking of trying docker but even then I'm not sure if that is a viable solution.

eenewbsauce, 2017-07-05:
I am running into a similar issue. I fired up a Docker container today and I'm running into more issues with 64bit kernel vs the 32bit of the Pi. I know it sounds silly but I'm going to try and code my prototype directly on the Pi for now. There are too many "software" things getting in the way of my "hardware" project.

eaglejs, 2017-07-05:
Yeah, I think that might make the most sense. I'll close this issue; hopefully, it will serve as info for anyone else.

closed 2017-07-05 by eaglejs

JamesBarwell, 2017-07-06:
Cheers for the report. Someone else who wanted to develop on Windows reported the same thing privately to me earlier this week, but I hadn't joined it up in my head with this issue.

I've added a note to the readme for the moment. I don't develop on Mac or Windows myself so I don't have any solution to recommend at the moment. Docker sounds promising. On the rare occasions when I get a chance to work on this, I tend to dev on the Raspberry Pi hardware itself, so that I can run the integration tests.

eaglejs, 2017-07-06:
Cheers!

I'll have to experiment and get back to you with Docker. I would develop on the Pi, but it's a little sluggish haha even with the extreme sd card. I'm writing a garage door opener application, and I already have it wired up to the garage since I had a prototype working with Python. But I wanted to give the MEAN stack a crack at utilizing the Pi. Python really isn't my strength, but I guess it is always an option as well. :) Anyways, I'll keep you posted if Docker might be a solution. There is an image out there that is supposed to be a flavor or raspbian. So we shall see. :)

commit fdf038be (2017-07-06): Add note regarding epoll Linux dependency. Fixes #67