JamesBarwell/rpi-gpio.js #74: Nothing happens.. it just hangs..

kdupreez, 2017-10-30:
I tried the sample below, but it logs to console "written pin" but then the app just hangs and I have to ctrl-C it.. nothing happens on the pins ether.

var gpio = require('rpi-gpio');

gpio.setup(7, gpio.DIR_OUT, write);

function write() {
    gpio.write(7, true, function(err) {
        if (err) throw err;
        console.log('Written to pin');
    });
}

thecodershome, 2017-10-30:
I'm not certain why nothing happens to the pins, but adding `process.exit()` after the console log should stop the process and prevent the need to Ctrl-C.
```js
var gpio = require('rpi-gpio');

gpio.setup(7, gpio.DIR_OUT, write);

function write() {
  gpio.write(7, true, function (err) {
    if (err) throw err;
    console.log('Written to pin');
    process.exit(0); // Stop the process
  });
}
```

kdupreez, 2017-10-31:
I gave up trying to use this and wrote a piece of code that does direct file writes to /sys/class/gpio/gpioX . 

that works great..

JamesBarwell, 2017-11-09:
Sounds like you're not intending to return to this code, and it's not something I could re-create, so I'll close this one.

For future reference, the code above doesn't check for errors coming from the `setup` method. It could have been that it was failing silently for some reason. Perhaps if there was an error it would have given more information.

closed 2017-11-09 by JamesBarwell