JamesBarwell/rpi-gpio.js #77: Node Issues

PSJMcNeill, 2017-12-03:
When running npm install I ran into problems with Node gyp build.

It seems it is due to new versions of Node using "nodejs" as the command.

Installing nodejs-legacy fixed the issue.

But may need added to the installation instructions or updated to use nodejs?

JamesBarwell, 2017-12-04:
Do you know what version of node you were using before installing nodejs-legacy?
Are you using Raspbian linux? Again, what version?

It doesn't sound like the kind of thing that could be fixed in this project but as you say perhaps the documentation could be updated.

PSJMcNeill, 2017-12-04:
Not too sure. I had a fresh installation of raspbian. So I assume the apt-get just got the latest version of Node.

Will have a look to see if I can find release notes of when that change happened.

JamesBarwell, 2018-11-09:
All the dependencies have been updated, and there's a guide now about how to run the latest now, so I'm going to close this off. Please create a new issue if you continue to have issues with the install. Cheers.

closed 2018-11-09 by JamesBarwell