JamesBarwell/rpi-gpio.js #79: write errno: -22

xyboox, 2018-02-10:
Hi,
I was trying to test the examples and seems like I'm stuck with this error:
`{ Error: EINVAL: invalid argument, write errno: -22, code: 'EINVAL', syscall: 'write' }`

Here's my code:
`// GPIO
    var gpio = require('rpi-gpio');
    
    var pin     = 4;
    var delay   = 2000;
    var countIO = 0;
    var max     = 3;

    gpio.setup(pin, 'out', on);

    function on (err) {
        if (err) {
            logger.info('GPIO Setup ERROR: %O', err);
            return;
        }
        if (countIO >= max) {
            gpio.destroy(function() {
                logger.info('Closed pins, now exit');
            });
            return;
        }

        setTimeout(function() {
            gpio.write(pin, 1, off);
            countIO += 1;
        }, delay);
    }

    function off() {
        setTimeout(function() {
            gpio.write(pin, 0, on);
        }, delay);
    }`

I'm using node v8.9.0 on a Raspberry Pi Zero

Any idea about what's wrong?
Thanks!

thecodershome, 2018-03-25:
Can you upload your debug log? Try adding `DEBUG=rpi-gpio` before your command

filiptc, 2019-01-06:
Same thing happening here. GPIO (pin 2) interaction is fine via BASH. Debug output:
```
> var pomise = gpiop.setup(2, gpio.DIR_OUT)
      .then(() => gpiop.write(2, false))
      .catch(e => { console.log(e); });
undefined
>   rpi-gpio seen hardware revision 8322; using pin mode v2 +0ms
  rpi-gpio set up pin NaN +8ms
  rpi-gpio export pin NaN +5ms
{ [Error: EINVAL: invalid argument, write] errno: -22, code: 'EINVAL', syscall: 'write' }
```