JamesBarwell/rpi-gpio.js #81: [v0.9.1] 'setup', 'read' and 'write' methods now require string instead of number

bellerofonte, 2018-04-05:
Hi!
Then I call following:
`GPIO.setup(13, GPIO.DIR_LOW, err => { .... });`
I get error:

```
TypeError: Cannot read property '13' of undefined
at getPinRpi (/opt/radiobox/node_modules/rpi-gpio/rpi-gpio.js:358:27)
```
and obviously there is an error:
```
function getPinRpi(channel) {
        return currentPins[channel] + '';    // <----- channel is Number
    };
```
and currentPins are:
```
v2: {
        // 1: 3.3v
        // 2: 5v
        '3':  2,
        // 4: 5v
        '5':  3,
        // 6: ground
        '7':  4,
        '8':  14,
        // ...etc
   }
```

The solution is:
```
function getPinRpi(channel) {
        return currentPins[`${channel}`] + '';    
    };
```

JamesBarwell, 2018-04-11:
I'm struggling to understand this one. I don't see why converting `channel` to a string before using it to lookup an object property would make any difference, because all object keys in JS are automatically converted to be strings anyway.

```js
> var foo = { '3': 2, '5': 3 };
undefined
> foo[3]
2
> foo['3']
2
```

The error you're getting looks to be due to `currentPins` being undefined rather than anything to do the `channel` value.

You started with "Then I call following". Is there some missing code that you run before this that could have affected the value of `currentPins`? Could there be an error occurring somewhere further up in your code that isn't being caught?

bellerofonte, 2018-04-12:
Hi!
This issue occurred after clean install of nodejs and all dependencies, inc. rpi-gpio.js
This code:
```
const GPIO = require('rpi-gpio');
GPIO.setup(13, GPIO.DIR_LOW, err => { .... });
```
 thrown error
```
TypeError: Cannot read property '13' of undefined
at getPinRpi (/opt/radiobox/node_modules/rpi-gpio/rpi-gpio.js:358:27)
```
The only change I have made was:
13 -> '13'
and 
```
const GPIO = require('rpi-gpio');
GPIO.setup('13', GPIO.DIR_LOW, err => { .... });
```
worked fine

bellerofonte, 2018-04-12:
Just tried to revert changes.
As you have expected, 
```
const GPIO = require('rpi-gpio');
GPIO.setup(13, GPIO.DIR_LOW, err => { .... });
```
now works fine too. 
Magic...

Maybe issue appeared after clean install and has gone since reboot or smth else.
Anyway, I cannot reproduce it anymore, so closing the issue. Thanks for your response!

closed 2018-04-12 by bellerofonte