JamesBarwell/rpi-gpio.js #82: Switch on and off a lamp using relay

muscaiu, 2018-05-13:
Is there an example for switching a simple lamp on and off using a relay?
I'm using a raspbery pi and a 5v relay, i'm able to switch it on and off using python code, but i'm not into python. I would like to use node and js instead.

**This is my setup:**
[image]

Thank you!

swee0720, 2018-07-05:
Could you post the python code for this particular setup please

fusionstream, 2019-03-17:
This isn't a support forum. It's a place to track bugs and feature requests for this module. There is an example for writing and reading the gpio pins on the main page of this doc.