JamesBarwell/rpi-gpio.js #85: change listener

demobiel, 2018-05-26:
Hello,

I have a weird thing that I don't really understand. My pins are connected to three buttons, each with pull up resistors.

this is my code:

gpio.setup(8, gpio.DIR_IN, gpio.EDGE_BOTH);
gpio.setup(10, gpio.DIR_IN, gpio.EDGE_BOTH);
gpio.setup(12, gpio.DIR_IN, gpio.EDGE_BOTH);

gpio.on('change',function(channel,value){
     console.log(channel,value);
})

if I press the button connected to pin 10, I get the false true for value, but the event is also called for the other channels with their current value. Is this normal?

result in terminal after clicking on button 10

10 false
12 true
8 true
12 true
10 true
8 true

Am I misunderstanding the library and is this normal behaviour? Or is there something wrong with my circuit giving false positives? On the other side I don't understand why the "change" listener would be called for the other channels as their values are not changing.

thanks!

JamesBarwell, 2018-05-27:
I'll just caveat this by saying I've not run the code myself...

It looks to me like you're getting 2 sets of 3 events. The reason for the two sets is because you're using `EDGE_BOTH`, so it's picking up the button down and up voltage changes. So if you changed it to `EDGE_RISING` I think that would fix part of it and you'd end up with just "10 false, 12 true, 8 true". If you wanted to turn off events for the other pins entirely then you could pass `EDGE_NONE`. Hopefully that's at least enough to help get around the problem.

I don't know why you're getting an event for each pin though. That does sound odd.

Before making any changes, could you try running the program with an environment variable set as in the example below, and give us the full output please? That'd be helpful just to confirm how the listener is being bound and to get an idea of whether it's something in this module or something lower down.

Like this:
```
DEBUG=rpi-gpio node yourapp.js
``

demobiel, 2018-05-31:
hi

I am running some tests, but sometimes it is there sometimes not... not sure :)

I hope to get somewhere with the debug tool!

jaspersorrio, 2020-04-30:
Hi @JamesBarwell awesome library. 

I am not sure why but I am getting weird results for the change handler. 

[image]