JamesBarwell/rpi-gpio.js #86: Node v10 Build Failing

williamluke4, 2018-06-13:
According to fivdi/epoll#27 this is because epoll need to be upgraded

```bash 
pi@raspberrypi:~/server $ npm install

> epoll@0.1.22 install /home/pi/server/node_modules/epoll
> node-gyp rebuild

gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make: Entering directory '/home/pi/server/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../../nan/nan.h:192:0,
                 from ../src/epoll.cc:15:
../../nan/nan_maybe_43_inl.h: In function ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’:
../../nan/nan_maybe_43_inl.h:112:15: error: ‘class v8::Object’ has no member named ‘ForceSet’
   return obj->ForceSet(isolate->GetCurrentContext(), key, value, attribs);
               ^
In file included from ../src/epoll.cc:15:0:
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’:
../../nan/nan.h:835:60: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated (declared at /home/pi/.node-gyp/10.4.1/include/node/node.h:171): Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, func, argc, argv);
                                                            ^
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’:
../../nan/nan.h:850:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated (declared at /home/pi/.node-gyp/10.4.1/include/node/node.h:164): Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, symbol, argc, argv);
                                                              ^
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’:
../../nan/nan.h:865:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’ is deprecated (declared at /home/pi/.node-gyp/10.4.1/include/node/node.h:157): Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, method, argc, argv);
                                                              ^
../../nan/nan.h: In member function ‘v8::Local<v8::Value> Nan::Callback::Call_(v8::Isolate*, v8::Local<v8::Object>, int, v8::Local<v8::Value>*) const’:
../../nan/nan.h:1479:5: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated (declared at /home/pi/.node-gyp/10.4.1/include/node/node.h:171): Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
     ));
     ^
epoll.target.mk:95: recipe for target 'Release/obj.target/epoll/src/epoll.o' failed
make: *** [Release/obj.target/epoll/src/epoll.o] Error 1
make: Leaving directory '/home/pi/server/node_modules/epoll/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/.nvm/versions/node/v10.4.1/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:237:12)
gyp ERR! System Linux 4.4.13-v7+
gyp ERR! command "/home/pi/.nvm/versions/node/v10.4.1/bin/node" "/home/pi/.nvm/versions/node/v10.4.1/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/server/node_modules/epoll
gyp ERR! node -v v10.4.1
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN server@1.0.0 No description
npm WARN server@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! epoll@0.1.22 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.22 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-06-13T01_53_41_156Z-debug.log
pi@raspberrypi:~/server $ 
```

JamesBarwell, 2018-06-13:
This is already being worked on here: https://github.com/JamesBarwell/rpi-gpio.js/tree/v2

I'll leave this open until the branch gets merged.

williamluke4, 2018-06-13:
Awesome thanks James

Erozionn, 2018-08-06:
Its still broken on 10.8.0. Not sure what to do..

JamesBarwell, 2018-08-06:
I've pushed another patch to the v2 branch to bump epoll again. I've installed dependencies then run the tests in a Docker container running node 10.8.0 and that all passes for me. Please could you try pulling that down and rebuilding? Let us know if it works, or if not please post the output.

jguillod, 2018-08-11:
But how should we install tree/v2 for node v10.8.0?
`npm i rip-gpio` points to main branch, isn't it?

JamesBarwell, 2018-08-13:
Yes, it still points to the main v1 branch.

Try this:
```
npm install git://github.com/JamesBarwell/rpi-gpio.js.git#v2 --save
```

williamluke4, 2018-08-13:
Seems to be working

JamesBarwell, 2018-11-09:
Version 2 has been merged to master and is published as the proper npm package now. Cheers.

closed 2018-11-09 by JamesBarwell