JamesBarwell/rpi-gpio.js #88: Pin write failing silently on conditional code

tiagodiogo, 2018-06-14:
**Specs**
Board: RaspberryPi 3
Distro: Raspbian Stretch
Node.js: v8.11.3

**Short Description**
The method gpio.write is not working inside conditional code (switch statements, if clauses)
Works normally on non conditional code

**Code Description**
The following code represents a web socket client that writes to a pin when receiving a message with certain parameters.
If the gpio.write is called before the conditional code (switch statement) the pin is correctly activated
If the gpio.write is called inside the conditional code no action happens

**Code Example**

`//StartUp the Socket Connection
ws.on('open', function open() {
  console.log("Socket Connection to Master Established");
  var msg = {
    type: MSGTYPE.PING,
    text: "Greetings Master",
    sender: hostname,
    date: Date.now(),
  };
   ws.send(JSON.stringify(msg));
});`

`//Incoming Message Handler
ws.on('message', function incoming(msg) {
  interpretMessage(ws, msg);
});`

`//Incoming Message Interpreter -> Evaluate and Respond
function interpretMessage(ws, msg){
  var message = JSON.parse(msg);
  console.log('Message | Type: %s | Sender: %s | Date: %s | IP: %s | Text: %s',
      Object.keys(MSGTYPE).find(key => MSGTYPE[key] === message.type), message.sender,
      new Date(message.date), ws._socket.remoteAddress, message.text);

 ## gpio.write(output, true); <- here works fine

  switch(message.type){
    case MSGTYPE.PING: {
      var msg = {
          type: MSGTYPE.ACK,
          text: "PING RECEIVED",
          sender: hostname,
          date: Date.now(),
        };
        ws.send(JSON.stringify(msg));
      break;
    }
    case MSGTYPE.ACTION: {
      var msg = {
          type: MSGTYPE.ACK,
          text: "ACTION RECEIVED",
          sender: hostname,
          date: Date.now(),
        };
      ws.send(JSON.stringify(msg));
      ## gpio.write(output, true); <- here nothing happens
      break;
    }
  };
};`

Can you please help me identifying something i am doing wrong or how i can make this work under conditional code?

Thank you,

Tiago

JamesBarwell, 2018-06-15:
I think the most likely scenario here is that your code isn't getting to the conditional bit that you think it is.

Try putting some `console.log` statements in your code just above the `write` commands, and log out the arguments you're passing in, to prove that the code is executing `write` correctly.

tiagodiogo, 2018-06-15:
hello Jason,
thanks you for your fast reply.
I understand that is a common bug but i am sure that the conditional code is being executed.
i have done many tests before posting the issue

tiagodiogo, 2018-06-15:
also, i have implemented the gpio.write callback and altough the PIN does not change its output the callback is called so the write function appears to be executing

JamesBarwell, 2018-06-15:
It's going to be hard to get much further with the snippet of code you've provided so far. There's no way I can re-create the bug from that.

If you are convinced that it's happening because of the call being made inside a conditional block of code, can you boil that down to a simple code example to help demonstrate that the issue is within this module?

You could also try running it with the `DEBUG=rpi-gpio` env var set and see if there's any clue in that output.

tiagodiogo, 2018-06-16:
i created the simple code example to demonstrate the problem and your library worked.
i don't know why it is not working in my example, it must be related to it being a web socket event but well, that is my problem to fix... 
i won't disturb you any longer, your code is definitely working.

thank you for your assistance

closed 2018-06-16 by tiagodiogo