JamesBarwell/rpi-gpio.js #89: Support for RaspberryPi 3B+ ?

iChUdY, 2018-08-13:
Hello,

not work on rpi3b+ :(

pi@raspberrypi:~ $ sudo npm install rpi-gpio

> epoll@0.1.22 install /home/pi/node_modules/epoll
> node-gyp rebuild

gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/node_modules/epoll/build'
gyp ERR! System Linux 4.14.52-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/epoll
gyp ERR! node -v v8.11.3
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! epoll@0.1.22 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.22 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-08-13T06_03_01_736Z-debug.log
pi@raspberrypi:~ $

hwangzhiming, 2018-08-13:
any plan for 3B+

denimaaark, 2018-10-11:
When will be supported 3B+ model please?

denimaaark, 2018-10-12:
Thank You, installation with instructions was successfull :)

closed 2018-11-09 by JamesBarwell