JamesBarwell/rpi-gpio.js #93: Hint: epoll Update regarding Windows.

Kassmadin, 2019-06-14:
Regarding the following statement:

> Please note that this module has a dependency on epoll and that currently it is only possible to build and develop the module on Linux systems.

Is just want to point out the following epoll issue
https://github.com/fivdi/epoll/issues/31

Maybe thats a "fix" for that?

Context: I also develope on a windows machine but can't use npm install rpi-gpio as its explodes with an error.

JamesBarwell, 2019-06-17:
Hi, thanks for pointing this out. I've pushed a branch `epoll-2.0.9` that bumps to this version, which includes that fix. Do you have time to give it a test please? I don't have time to get a Windows system set up for dev at the moment.

```
npm install git://github.com/JamesBarwell/rpi-gpio.js.git#epoll-2.0.9 --save
```

Another possible option (not one I've tried myself) is to use a docker-based dev environment running on Windows (or other platforms!).

Kassmadin, 2019-06-17:
Sure I will as soon as I have time

Kassmadin, 2019-06-22:
Might take a little longer.

tornordqvist, 2019-07-06:
epoll@2.0.9 install correctly on Windows 10, just tested it.

But I have problem installing latest version of rpi-gpio on Raspberry/Raspbian/Armv71:

Looks like this:
```
pi@raspberrypi:~/repos/rpi-led-local-server-nodejs $ npm i rpi-gpio

> epoll@0.1.22 install /home/pi/repos/rpi-led-local-server-nodejs/node_modules/rpi-gpio/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/repos/rpi-led-local-server-nodejs/node_modules/rpi-gpio/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../../nan/nan.h:192:0,
                 from ../src/epoll.cc:15:
../../nan/nan_maybe_43_inl.h: In function ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’:
../../nan/nan_maybe_43_inl.h:112:15: error: ‘class v8::Object’ has no member named ‘ForceSet’
   return obj->ForceSet(isolate->GetCurrentContext(), key, value, attribs);
               ^~~~~~~~
In file included from ../src/epoll.cc:15:0:
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Valu
```

JamesBarwell, 2019-07-10:
Thanks for testing.

I'm a bit confused - from your output there, it looks like it's trying to install an older version of epoll. I can't see which version of node you're on, but we need to be on epoll 2.x to work in node 10 (see: https://github.com/fivdi/epoll/issues/27). Do you have a package.json in there referencing the old 1.x version perhaps?

I just tried it myself, running `npm i rpi-gpio` in a clean directory, and it installs epoll 2.0.9.