JamesBarwell/rpi-gpio.js #98: This library supported raspberry 4?

psyrenpark, 2019-11-30:
This library supported raspberry 4? 

I want use gpio library.

JamesBarwell, 2019-12-02:
I expect it will work fine on the Pi 4, but I've not got one myself yet so haven't been able to confirm. If you have a chance to give it a test, please let us know. Cheers.

Roaders, 2020-05-07:
I've tested on a raspberry pi 4 and it works fine.

hartleyrobertson, 2020-05-08:
I can also confirm that it works on pi 4. Not sure if this is useful to anyone but no other version of NodeJs worked except for 10.20.1. Kept getting an install error relating to epoll.

JamesBarwell, 2020-05-20:
Thanks for testing. I'll add it to the compatibility list. Not sure about the node version issue but we could get that raised as a separate issue.

commit 34622833 (2020-05-20): Add RPI 4 B to compatibility list. Fixes #98
closed 2020-05-20 by JamesBarwell