JamesBarwell/rpi-gpio.js #100: Convert library to typescript

Roaders, 2020-05-07:
not really an issue more of a question. I am going to be using this library to create a node / nestjs based garage door opener and my first step will be to write declaration files for it so that I can use it in a Typescript project.

Converting the project to Typescript is actually easier than writing declaration files and it means that the declarations do not become out of date.

Would you accept a PR converting this to typescript? it wouldn't affect the way it is consumed at all.

I could of course clone and release an identical package with types but that will further clutter npmjs!

I will start the conversion and submit a PR. IF you merge it great, if not I'll release a ts based version of this library.

JamesBarwell, 2020-05-20:
Firstly I'll start by saying that I use TS every day professionally. But I'm not minded to switch this project over to it, at least not any time soon. What I've learned since publishing this module many years ago is that a significant number of people using it are fairly new to programming, or new to JS. Some of the contributions to this project have been from people who are not used to contributing to open source software, and this may well be their first time.

Switching to TS will raise the barrier of entry and I'm not comfortable with that, at least at the moment when most hobby developers are not using TS. The other reason is that it's a small module with no complex types, so I'm not convinced that TS has much to offer here - the behavioural tests are far more valuable.

I don't worry too much about a separately hosted definition file going out of date. I have taken compatibility very seriously over the years, and the interface has been pretty stable and small.

Yes please do go ahead and publish a TS fork if you think it will offer some benefits.

closed 2020-05-20 by JamesBarwell

JamesBarwell, 2020-05-21:
p.s. someone already wrote some definitions in #73 but it doesn't look like they bothered to put them on Definitely Typed.

Roaders, 2020-05-21:
the types are now available on Definitely Typed.

JamesBarwell, 2020-05-21:
Great stuff. I've added a note to the readme.