JamesBarwell/rpi-gpio.js #109: gpio.accessible() method

jjeff, 2020-09-13:
I'm creating a cross-platform Node app which uses GPIO pins when they're available. However, on other platforms, I'd like to save the overhead (and errors) of the rpi-gpio calls.

Would you be interested in a pull request for a method which would return false if GPIO is not available?

JamesBarwell, 2020-09-29:
Hi, I've been having a think about it. On one hand it seems like a huge edge-case and not something others will use, but on the other there is already some RPI platform detection code in the module (`setRaspberryVersion`), so perhaps there's a neat way to expose this info without adding much complication.

Did you have an approach in mind for detecting whether the GPIOs are available? What do you think about just exposing this already-collected RPI platform info?

jjeff, 2020-09-29:
Here's what I'm currently doing:

``` javascript
function isGpioAccessible() {
  let fd;
  try {
    fd = fs.openSync('/sys/class/gpio/export', 'r+');
  } catch(e) {
    return false;
  } finally {
    if (fd) {
      fs.closeSync(fd);
    }
  }
  return true;
}
```

JamesBarwell, 2020-09-30:
Cool, yeah it's simple enough and fairly self-contained. Yeah please open a PR and I'll try and get it in soon.