JamesBarwell/rpi-gpio.js #119: Segmentation Error on instantiation (v2.1.7) on Pi OS

crossoft, 2022-06-27:
user@raspberrypi:~/gpio-controller $ node
Welcome to Node.js v16.15.1.
Type ".help" for more information.
> var gpio = require("rpi-gpio")
Segmentation fault

Pi info:

user@raspberrypi:~/gpio-controller $ uname -r
5.15.32-v7+
user@raspberrypi:~/gpio-controller $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

crossoft, 2022-06-27:
I'll leave this here in case it's beneficial, but I decided just to rerun `npm i` and all worked.  First time in 50+ pi's that I've used this on.  Thanks!

JamesBarwell, 2022-09-07:
Not sure what happened there, but glad you were able to resolve it. Thanks for reporting it.

closed 2022-09-07 by JamesBarwell